badgeteam/ESP32-platform-firmware #52: Campzone: rgb.pixel only adds

sebastius, 2019-07-21:
rgb.pixel(255,255,255,0,0) works
rgb.pixel(0,0,0,0,0) after that doesnt and requires a clearing of the display

Duality4Y, 2019-08-26:
So what would be the correct behavior?

When i look at the code i see that indeed the colors are added together.
But are the colors supposed to be set to 0 when you call say rgb.pixel((0, 0, 0), (0, 0))?

What i see is that the rgb code uses the compositor and that the compositor layers all this pixel data together by adding them together.

Something i noticed i could do is something like this:

``` python
rgb.pixel((0xff, 0xff, 0xff), (0, 0)) # turns on pixel at 0, 0
rgb.pixel((-0xff, -0xff, -0xff), (0, 0)) # turns off pixel at 0, 0
```

renzenicolai, 2019-08-27:
The RGB API will be replaced by the generic display api soon.

In fact: try this on a campzone badge:

rgb.disablecomp()
display.drawLine(0,0,10,10,0xFF0000)
display.flush()

:-)

closed 2019-08-27 by renzenicolai