badgeteam/ESP32-platform-firmware #77: "OSError: 257" when activating wifi from dashboard on SHA2017 badge

raboof, 2019-08-13:
on serial console:
```
Traceback (most recent call last):
  File "boot.py", line 26, in <module>
  File "dashboard/home.py", line 266, in <module>
  File "wifi.py", line 21, in connect
OSError: 257
```

wifi.py is:
```
def connect(ssid=defaultSsid, password=defaultPassword):
        global sta_if
        if not ssid:
                return status()
        sta_if.active(True)
        print("WiFi connect to",ssid,password)
```

line 21 is `sta_if.active(True)`

(running version ca1dc5c)

raboof, 2019-08-13:
Getting the same error when opening the wifi settings app:
```
Traceback (most recent call last):
  File "boot.py", line 26, in <module>
  File "dashboard/settings/wifi.py", line 16, in <module>
OSError: 257
```

That line is:
```
sta_if = network.WLAN(network.STA_IF); sta_if.active(True)
```

raboof, 2019-08-13:
From the REPL:
```
E (79845) wifi: Expected to init 10 rx buffer, actual is 8
E (79855) [modnetwork]: Error initializing WiFi (257)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
```

raboof, 2019-08-13:
That error appears to be thrown from libpp.a, filed https://github.com/espressif/esp32-wifi-lib/issues/21 on the off-chance that they might be able to help further narrow down the problem.

renzenicolai, 2019-08-13:
It works on my badge.

Please execute the following commands:
cp firmware/configs/sha2017_defconfig firmware/sdkconfig
./build.sh
./flash.sh

and try again

----------------

BADGE.TEAM
>>> import wifi
>>> wifi.connect()
WiFi connect to <REDACTED>
>>> wifi.status()

renzenicolai, 2019-08-13:
I also tried connecting while the splash screen is running: works just fine.

raboof, 2019-08-14:
> It works on my badge.
> I also tried connecting while the splash screen is running: works just fine.

I don't doubt it! Apparently there's _something_ different on my end...

> Please execute the following commands

Pretty sure that's how I built and flashed mine - I'll try a fresh checkout later.

raboof, 2019-08-14:
> Please execute the following commands (...) and try again

Still the same behaviour

>import wifi
>wifi.connect()
>E (27256) wifi: Expected to init 10 rx buffer, actual is 8
>E (27266) [modnetwork]: Error initializing WiFi (257)
>Traceback (most recent call last):
>  File "<stdin>", line 1, in <module>
>  File "wifi.py", line 21, in connect
>OSError: 257

renzenicolai, 2019-08-14:
Can you try flashing the firmware binary provided here? https://badge.team/firmware/sha2017.bin

Just to make sure that the source of the problem lies in the compilation process and not in hardware.

Which toolchain do you use?
Which IDF do you use? The IDF included with this repository (this version is used when ./build.sh is used, or when running "export IDF_PATH=../esp-idf" and running make from the firmware folder)?
Do you use the default menuconfig for sha2017 badges while building?

raboof, 2019-08-14:
> Can you try flashing the firmware binary provided here? https://badge.team/firmware/sha2017.bin
> Just to make sure that the source of the problem lies in the compilation process and not in hardware.

Flashed with `./esp-idf/components/esptool_py/esptool/esptool.py write_flash 0x10000 ~/Downloads/sha2017.bin`

The problem persists...

Full output when opening the wifi setup app:

```
OTA1 is at 0x00190000
Unpacking ZIP...
E (1319) ZIP: no preseed .zip found
Starting app 'dashboard.settings.wifi'...
E (3449) wifi: Expected to init 10 rx buffer, actual is 6
E (3459) [modnetwork]: Error initializing WiFi (257)
Fatal exception in the running app!
Failed to load image.
Error in PNG draw Rendering error
Traceback (most recent call last):
  File "boot.py", line 26, in <module>
  File "dashboard/settings/wifi.py", line 16, in <module>
OSError: 257
```

> Which toolchain do you use? Which IDF do you use? The IDF included with this repository (this version is used when ./build.sh is used, or when running "export IDF_PATH=../esp-idf" and running make from the firmware folder)?

The linux one from the repo, with `./build.sh`

> Do you use the default menuconfig for sha2017 badges while building?

The one from `firmware/configs/sha2017_defconfig`, yes

renzenicolai, 2019-08-14:
Very strange...

That binary is confirmed to be working on other badges. Have you tried erasing the flash? (./erase.sh) and then flashing again (/./flash.sh)?

Will you visit CCC camp next week? If so we could try to figure things out then :-)

raboof, 2019-08-14:
> Have you tried erasing the flash? (./erase.sh) and then flashing again (/./flash.sh)?

Ugh, that did the trick... thanks a lot for the help!

> Will you visit CCC camp next week?

Unfortunately no! Have fun there!

closed 2019-08-14 by raboof