badgeteam/ESP32-platform-firmware #83: make ./flash.sh flash both OTA partitions

raboof, 2019-08-18:
The badge currently has 2 firmware partitions: one for the 'last working' firmware and one to download a new OTA update into.

`./flash.sh` currently only flashes one, so you have a 50/50 change of actually seeing your changes. As a workaround you can do a firmware update :).

It would be nice if `./flash.sh` flashed both firmware partitions

raboof, 2019-08-18:
alternatively I guess it could reset the field that determines from which partition to boot

renzenicolai, 2019-08-18:
I've added a script to erase the bootloader configuration partition. This resets the used OTA partition to partition 1. By the way: this partition selection thing is all the configuration partition stores. I don't find it usefull at this time to look at the exact binary contents and provide a method of switching to OTA2 on demand.

closed 2019-08-18 by renzenicolai