badgeteam/ESP32-platform-firmware #84: serial menu's clear the console

raboof, 2019-08-18:
The menu system on the serial is cool, but if you're debugging an app it also thends to clear the screen when you're trying to digest a stack trace :D.

Perhaps we should have an option to enable/disable it?

renzenicolai, 2019-08-18:
Done

raboof, 2019-08-19:
It's not quite clear to me what was changed to resolve this issue? I don't see an option to enable/disable it and it still appears to clear the screen.

raboof, 2019-08-19:
Apparently there is now a 'settings -> homescreen configuration' setting, but that doesn't quite seem to work for me: disabling it indeed gives me a cleaner serial, but also the badge doesn't seem to respond to touch anymore.

renzenicolai, 2019-08-19:
Bug found, buag fixed in pr https://github.com/badgeteam/ESP32-platform-firmware/pull/93

closed 2019-08-19 by renzenicolai

raboof, 2019-08-19:
Confirmed fixed