badgeteam/ESP32-platform-firmware #85: serial communciation with mpfshell does not work

raboof, 2019-08-18:
connecting with mpfshell (for file transfer and REPL) doesn't work, presumably because of the nice fancy serial menu system.

Is there a way we can have both?

renzenicolai, 2019-08-18:
Try again now.

You can also leave the menu enabled now. Should not make a difference as it will now reboot to a shell when it detects CTRL+C or CTRL+D

raboof, 2019-08-19:
Indeed on opening the serial connection with mpfshell now triggers the application shell app on the badge, though the first time entering the REPL doesn't work:

```
 SHA2017 - Loading application shell...

ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:4024
load:0x40078000,len:8112
ho 0 tail 12 room 4
load:0x40080000,len:5988
entry 0x400802c4

BADGE.TEAM
Welcome to the python shell of your badge!
Check out https://wiki.badge.team/MicroPython for instructions.
>>>

could not enter raw repl

mpfs [/]>

```

... but the second time it does:

```
mpfs [/]> repl
>
*** Exit REPL with Ctrl+] ***

>>> import badge
>>> badge.
__class__       __name__        consts          display
eink            init            machine         mpr121
neopixel        nvs_get_u16     nvs_get_u8      nvs_set_u16
nvs_set_u8      time            setPower        eink_busy_wait
usb_volt_sense  safe_mode       png_info        png
nvs_set_str     nvs_get_str     deviceType      _vbat
_vusb           eink_init       eink_png        leds_init
leds_enable     leds_send_data  battery_volt_sense
>>> badge.
```

Seems sufficient for now :+1:

closed 2019-08-19 by raboof