badgeteam/ESP32-platform-firmware #97: Add an option reset the badge back to factory defaults via UI

jhaand, 2019-08-21:
Add an option to reset the badge back to factory settings from the UI. 

At our hackerspace we use HackerHotel badge for instruction. People learn to hack their way through firmware. like setting up the wifi, install apps and such. It would be nice to reset the badge back to basics after the people are done with them. 

info: https://tdvenlo.nl/wiki/doku.php?id=hh19_badge

raboof, 2019-08-21:
> At our hackerspace we use HackerHotel badge for instruction. People learn to hack their way through firmware. like setting up the wifi, install apps and such

Cool!

> It would be nice to reset the badge back to basics after the people are done with them.

Would `./erase.sh` do the trick?

renzenicolai, 2019-08-27:
./erase.sh
./flash.sh
would indeed do the trick, but I prefer having a proper reset function in the firmware. Added to the to-do list ;-)

renzenicolai, 2020-01-01:
Covered by #104

closed 2020-01-01 by renzenicolai