badgeteam/ESP32-platform-firmware #102: ugfx compatibility layer: set_default_font and char not implemented

niekproductions, 2019-08-26:
`ugfx.char(x, y, char, font, colour)` prints a single character.

`ugfx.set_default_font(font)` sets.. the default font. 

Both should be implemented in the compatibility layer.

renzenicolai, 2020-02-21:
While I do agree that these should be added, I also think we should start thinking about removing all references to ugfx from the firmware. As we're no longer using ugfx and ugfx is a brand name.

For this I shall start with moving the listbox emulation away from the ugfx compat library. Existing apps can be updated when we remove the ugfx layer completely.

renzenicolai, 2020-03-04:
UGFX compatibility layer will be completely removed soon. If there is any functionality UGFX had that you're missing in the platform firmware then let me know!

Note: I will take a look at adding a "set default font" feature.

closed 2020-03-04 by renzenicolai