badgeteam/ESP32-platform-firmware #103: flash serial port autodetection

raboof, 2019-08-28:
since 5462dc0e, the device to use for flashing is determined by:
```
CONFIG_ESPTOOLPY_PORT="`ls /dev/tty{USB0,.wchusbserial*,.SLAB_USBtoUART,ACMx} 2>/dev/null`"
```
This does not work for me (Debian): it seems the `{}` and the `*` wildcard don't seem to interact nicely, causing the expression not to return _any_ lines, not even /dev/ttyUSB0 .

/cc @tjclement

tjclement, 2019-08-28:
That's interesting! This was done to support flashing many badges at once during the packing sweatshop, which was ran on Ubuntu (surprising, because it is a Debian derivative) and Fedora systems. Do you know of alternative ways on Debian?

raboof, 2019-08-28:
wait, the backticks don't seem to be interpreted at all even... weird, investigating more

raboof, 2019-08-28:
do you happen to know which `make` you were using on that machine? GNU Make 4.2.1 here.

raboof, 2019-08-28:
It seems to not like the accolades notation here... what is your `dpkg -S /bin/sh`? also `dash`? `apt-cache policy dash` also `0.5.10.2-5`?

renzenicolai, 2019-11-18:
Removed this autodetect thing. It doesn't work on Linux, defeating it's own main purpose of allowing easy use of the software on both mac and linux.

closed 2019-11-18 by renzenicolai

tjclement, 2019-11-18:
I will look into this in the future. As my message above states, it does (did?) work on both standard Ubuntu desktop and Fedora, so it is incorrect to say it doesn't work on Linux.

It's a very handy feature for people new to embedded software development, so I see a lot of value to reimplement it later on.