badgeteam/ESP32-platform-firmware #107: CampZone rgb API needs to be enabled explicitly?

raboof, 2019-09-03:
I did a clean erase+flash of the campzone badge, which left me with a blank display.

Apparently it needs an explicit `rgb.enablecomp()` to enable the rgb module, which is still used by the campzone factory checks. After enabling this I could complete the factory check, but after that was back to a blank display - presumably because all the other apps have the same problem.

raboof, 2019-09-06:
The background here is that the rgb module is being phased out in favor of the display module. perhaps we should `rgb.enablecomp()` when loading the rgb module, that way apps using it will still work while new apps don't need to do anything to use the new display api

renzenicolai, 2019-11-18:
Should be somewhat fixed.

closed 2019-11-18 by renzenicolai