badgeteam/ESP32-platform-firmware #150: Missing media/{busy,alert}.png, dashboard.terminal.launcher (and possibly other files) on SHA2017 badge

zakx, 2020-01-01:
I have just pulled a SHA2017 badge from storage, which still had some on-event firmware on it (sadly I didn't check). I did the OTA update to "Happy new year", but almost everything crashes, complaining about missing files or modules.

Choosing "apps" from the serial menu results in this traceback:
```
Starting app 'dashboard.terminal.launcher'...
Traceback (most recent call last):
  File "boot.py", line 41, in <module>
ImportError: no module named 'dashboard.terminal.launcher'
Fatal exception in the running app!
Error in PNG draw Could not open file '/media/alert.png'!
```

Tools > File downloader results in:
```
Starting app 'dashboard.terminal.downloader'...
Traceback (most recent call last):
  File "boot.py", line 41, in <module>
  File "dashboard/terminal/downloader.py", line 5, in <module>
TypeError: unexpected keyword argument 'showStatus'
Fatal exception in the running app!
Error in PNG draw Could not open file '/media/alert.png'!
``` 

I'll now try to do a complete flash, but I suppose this should be working?

zakx, 2020-01-01:
A complete reflash of master did not fix this.

I was able to manually add the images by running `os.mkdir("media")` and https://raw.githubusercontent.com/badgeteam/ESP32-platform-firmware/e195b242fc87e29744406064d757cb13eea7a3f7/icons/debug-dump-icons.py in the REPL; however that doesn't supply missing modules like `dashboard.terminal.eink`. Any hints?

raboof, 2020-01-02:
> A complete reflash of master did not fix this.

A bit of a shot in the dark, but perhaps you need to run `reset_bl_to_ota1.sh` to make sure you're actually booting into the newly-flashed version and not a previous OTA update?

renzenicolai, 2020-01-02:
- Fixed file download utility
 - Created new launcher app with terminal menu

The icons should be present after the upgrade and definitely after a full wipe and reinstall.

try running this in the shell of your badge to manually install the icons:
`import dashboard.resources.png_icons`

closed 2020-02-13 by renzenicolai