badgeteam/ESP32-platform-firmware #172: Add flashing documentation

Zokol, 2020-01-31:
Erase the badge with "python2 esp-idf/components/esptool_py/esptool/esptool.py --port /dev/ttyUSB0 erase_flash", followed by this one to flash everything:

python2 esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 2000000 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 firmware/build/bootloader/bootloader.bin 0x8000 firmware/build/disobey2020_4MB.bin 0x10000 firmware/build/firmware.bin 0x10000 firmware/build/firmware.bin 0x191000 initial_files_disobey2020.zip

Note: to build a partition table .bin file select that table in menuconfig, then build the fw.

Note: this uses the existing 4MB partition table. The flasher should state the detected flash size. If this is 16MB then please try doing the same thing with the "sha2017_16MB.csv" partition table selected, this will increase the FAT partition to use the full flash size

Zokol, 2020-01-31:
```
~/ESP32-platform-firmware$ python2 esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 2000000 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 firmware/build/bootloader/bootloader.bin 0x8000 firmware/build/disobey2020_4MB.bin 0x10000 firmware/build/firmware.bin 0x10000 firmware/build/firmware.bin 0x191000 initial_files_disobey2020.zip
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'firmware/build/disobey2020_4MB.bin'
```

Zokol, 2020-01-31:
Used firmware/build/sha2017_16MB.bin instead, flashing process was completed, but badge crashed on boot.

Zokol, 2020-01-31:
@renzenicolai is the ota_data_initial.bin required at all? Previously it was flashed to the badge, but it is not included in the script above?

renzenicolai, 2020-01-31:
@Zokol The "ota_data_initial.bin" file makes sure the bootloader uses OTA1 to boot from, as OTA2 does not contain valid firmware yet. Erasing the OTA-data partition has the same effect but this is the more "official" route :P