badgeteam/ESP32-platform-firmware #188: Bluetooth support for the python API

mkorkalo, 2020-02-16:
It would be cool to have bluetooth support on the API.
Tons of things you could implement with it.
Upstream micropython has support: https://github.com/icetomcat/micropython/blob/master/ports/esp32/bluetooth/bluetooth.c

Lets say if I wanted to port this to ESP32-platform-firmware, is there anything special that should be taken into account? Also see my other issue regarding the build system & syncing upstream.

kliment, 2020-02-16:
The main issue with the bluetooth stack used there is that it has a very large code size and memory footprint. This is the reason we don't use it (even though many people want it). If you manage to make it work, awesome, but last we looked at it it wouldn't really be able to run at the same time as all the other functionality we have. It might be easier to port a different bluetooth stack meant for smaller devices if you can. For example, you look at the Arduino ESP32 Bluetooth implementation. There is a further issue in that the ESP32 Bluetooth hardware is not great in terms of performance or power consumption. Particularly the BLE hardware is really power-inefficient (worse than wifi iirc). So feel free to attempt it, but you might be disappointed with the hardware performance once you do.

renzenicolai, 2020-08-09:
Thanks to @jorisplusplus we now have a base for creating a new version of the firmware using the latest ESP-IDF and Micropython versions. Bluetooth will hopefully be a possibility there, but for this version of the platform adding bluetooth will not be possible. I'm closing this issue, but we do keep having bluetooth support in mind for the future :smiley:

closed 2020-08-09 by renzenicolai