badgeteam/ESP32-platform-firmware #211: Trying to build: implicit declaration of function 'gpio_install_isr_service'

n0p, 2020-05-18:
Relevant output:
```
Toolchain path: /home/n0p/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin//xtensa-esp32-elf-gcc
Toolchain version: esp-2020r1
Compiler version: 8.2.0
Python requirements from /home/n0p/src/micro/esp32/esp-idf/requirements.txt are satisfied.
Building partitions from /home/n0p/src/micro/esp32/badge-firm/firmware/partitions/sha2017_16MB.csv...

App "firmware" version: cz19v25-83-g5fd261be-dirty
CC build/main/platform.o
/home/n0p/src/micro/esp32/badge-firm/firmware/main/platform.c: In function 'isr_init':
/home/n0p/src/micro/esp32/badge-firm/firmware/main/platform.c:7:19: error: implicit declaration of function 'gpio_install_isr_service' [-Werror=implicit-function-declaration]
   esp_err_t res = gpio_install_isr_service(0);
                   ^~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [/home/n0p/src/micro/esp32/esp-idf/make/component_wrapper.mk:292: platform.o] Error 1
make: *** [/home/n0p/src/micro/esp32/esp-idf/make/project.mk:635: component-main-build] Error 2
```

robotman2412, 2020-08-02:
What badge?
Which configuration did you copy, if any?
It's usually helpful to provide information about configuration.

This issue is likely caused by not copying the sdkconfig for your badge.

tjclement, 2020-08-02:
It looks like this is using the latest xtensa toolchain instead of the one bundled with our firmware repo. Also looks like an old issue report we didn't see before; is it still open @n0p?

renzenicolai, 2020-08-03:
Yup, use the toolchain included with the repo, we're still using an older ESP-IDF version as updating that isn't plug-and-play.

closed 2020-08-03 by renzenicolai